MyLab-odyssey/ED_BMSdiag #10: Smart ED seems to keep resetting

casainho, 2017-05-10:
Hi. Thank you for making this project!!

I put and Arduino Pro Mini + a cheap MCP2515 together. The first issue I had was because I wired CAN_H and CAN_L exchanged -- I verified with oscilloscope and corrected. Now I connect on my Smart ED and the dashboard seems to keep reseting but at sometime says something like "workshop fault brake" or such. Any ideas what could be wrong? The signals of CAN_H e L seems to be ok.

This is the output:

[03:03:05:699] --- ED Battery Management Diagnostics ---
[03:03:05:711] --- v0.5.6                            ---
[03:03:05:711] -----------------------------------------
[03:03:05:711] Connect to OBD port - Waiting for CAN-Bus 
[03:03:05:724] .CONNECTED
[03:03:06:709] -----------------------------------------
[03:03:12:715] Reading dataF#0F#1F#2F#3F#4F#5F#6F#7
[03:03:42:764] -----------------------------------------
[03:03:42:773] Time [hh:mm]: 00:00,   ODO : 0 km
[03:03:42:773] -----------------------------------------
[03:03:42:786] SOC : 0.0 %, realSOC: 0.0 %
[03:03:42:786] HV  : 0.0 V, 0.00 A, 0.00 kW
[03:03:42:786] LV  : 0.0 V
[03:03:42:786] -----------------------------------------
[03:03:42:786] ENTER command... (? for help)

MyLab-odyssey, 2017-05-10:
Hi casainho!

Please give some more detail about the hardware of your project. 
a) There are two versions of the Arduino Pro mini - one 8MHz with 3.3V and one with 5V @ 16MHz. Which one do you use?
b) What is the CAN transceiver you use. You can't connect the Output of the MCP2515 directly to the car OBD port!!! Please use a MCP2551 when using a 5V MCU or a MCP2562 on 3.3V systems.
c) Did you follow the connection diagram correctly (for the OBD port / diagnostics port)?
d) Please use a example sketch and test for correct initialization of the MCP2515 CAN controller.

casainho, 2017-05-10:
Thank you for help me.

Please give some more detail about the hardware of your project.
> a) There are two versions of the Arduino Pro mini - one 8MHz with 3.3V and
> one with 5V @ 16MHz. Which one do you use?
>
5V 16MHz.

> b) What is the CAN transceiver you use. You can't connect the Output of
> the MCP2515 directly to the car OBD port!!! Please use a MCP2551 when using
> a 5V MCU or a MCP2562 on 3.3V systems.
>
Arduino CAN Bus Module Pin Outs and SchematicsMCP2515 TJA1050 CAN Bus
ModuleThese
Arduino CAN Bus modules are based on the MCP2515 CAN Controller and the
TJA1050 CAN Transceiver. They provide a lot of bang for the buck and will
allow you start to start interfacing your projects to CAN networks. Common
CAN applications include vehicles (via OBDII or J1939) and industrial
monitoring and controls.
Communication to this module with your Arduino is achievable via SPI. The
device includes a jumper configurable 120 Ohm Resistor and a CAN V2.B
support. More, there are existing libraries that make building projects
relatively easy.
Getting OneThese devices are readily available and can cost less than $2.00
including shipping.
Amazon
<https://www.amazon.com/gp/search/ref=as_li_qf_sp_sr_il_tl?ie=UTF8&tag=leaacicarbatf-20&keywords=Arduino%20CAN%20Bus%20Module&index=aps&camp=1789&creative=9325&linkCode=xm2&linkId=e560efc64860aaff56565574d24da4ef>
eBay
<http://rover.ebay.com/rover/1/711-53200-19255-0/1?icep_ff3=9&pub=5575103433&toolid=10001&campid=5337702195&customid=&icep_uq=Arduino+CAN+Bus+Module&icep_sellerId=&icep_ex_kw=&icep_sortBy=12&icep_catId=&icep_minPrice=&icep_maxPrice=&ipn=psmain&icep_vectorid=229466&kwid=902099&mtid=824&kw=lg>
Bang Good
<http://www.shareasale.com/r.cfm?u=1129382&b=305177&m=32599&afftrack=&urllink=www%2Ebanggood%2Ecom%2Fsearch%2Fcan%2Dbus%2Dmodule%2Ehtml>
Deal
Extreme
<http://www.shareasale.com/r.cfm?u=1129382&b=302497&m=32431&afftrack=&urllink=www%2Edx%2Ecom%2Fs%2FCAN%2BBus%2BModule>
IC
<http://www.shareasale.com/r.cfm?u=1129382&b=497343&m=48335&afftrack=&urllink=www%2Eicstation%2Ecom%2Fadvanced%5Fsearch%5Fresult%2Ephp%3Fkeywords%3Dcan%2Bbus%26search%5Fin%5Fdescription%3D1>
Station
<http://www.shareasale.com/r.cfm?u=1129382&b=497343&m=48335&afftrack=&urllink=www%2Eicstation%2Ecom%2Fadvanced%5Fsearch%5Fresult%2Ephp%3Fkeywords%3Dcan%2Bbus%26search%5Fin%5Fdescription%3D1>

> c) Did you follow the connection diagram correctly (for the OBD port /
> diagnostics port)?
>
Yes, I use 3 pins and verified on my oscilloscope (but without connection
to OBD port) this signals:
https://github.com/MyLab-odyssey/ED_BMSdiag/wiki/Troubleshooting-and-Tips#check-can-levels-if-necessary

> d) Please use a example sketch and test for correct initialization of the
> MCP2515 CAN controller.
>
I will try and give feedback.

Options for testing:
- verify if the 8MHz crystal is an issue
- should or should not connect GND on the OBD connectors? - this image for
instance, says no:
https://www.14core.com/wp-content/uploads/2015/12/OBD-Arduino-Interface-Connection.jpg

-- 
Cumprimentos,
Jorge Pinto
E-mail: casainho@gmail.com
Telemóvel: 927145983

MyLab-odyssey, 2017-05-10:
a) "verify if the 8MHz crystal is an issue" - this is a good point! 

I initialize the MCP2515 with 16 MHz. Please take a look at 8 MHz definitions in the [mcp_can_defs.h](https://github.com/MyLab-odyssey/ED_BMSdiag/blob/master/libraries/MCP_CAN_lib_15/mcp_can_dfs.h) and in the ED_BMSdiag project in the [canDiag.cpp](https://github.com/MyLab-odyssey/ED_BMSdiag/blob/master/ED_BMSdiag/canDiag.cpp) (CAN bus in the car is running at 500kbps):
`// Initialize MCP2515 running at 16MHz with a baudrate of 500kb/s and the masks and filters enabled.
  if(myCAN0->begin(MCP_STD, CAN_500KBPS, MCP_16MHZ) == CAN_OK) DEBUG_UPDATE(F("MCP2515 Init Okay!!\r\n"));
  else DEBUG_UPDATE(F("MCP2515 Init Failed!!\r\n"));`

I have never used the MCP2515 with 8 MHz and the library was not tested with this clock rate!

b) I always connected GND and CAN-L / CAN-H, because the driver stage on my transceivers need this.

Please report...

casainho, 2017-05-10:
Can you please tell me If I can change any define and get a kind of dmesg,
first to see the correct initialization of the MCP2515 CAN controller and
after the log of messages with the SmartED?

MyLab-odyssey, 2017-09-01:
This CAN board was verified to work, when initialized correctly with 8 MHz and used on the 3rd Gen. Smart ED.

closed 2017-09-01 by MyLab-odyssey

casainho, 2017-09-01:
Sad I couldn't make it working in my SmartED -- great the help I got from
you, really good work. I wish I can back later to test again. You are
developing this project in a very professional way -- CONGRATULATIONS!!

MyLab-odyssey, 2017-09-01:
Hello Jorge,

thanks for the completemts! Sad to read, that it is still not working for you. Please get the recommended hardware and test for yourself. Then build your compact version and test the CAN bus with small send- / receive-sketches between the two boards. I also spend hours checking CAN communication and did a lot of tests before I went to the car. When the communication is working proceed to the car. Do you really have a 3. gen Smart ED or one of those Car2Go 2. gen versions with Tesla-hardware in it? 

The project is now finished and I will release version 1.0 in some days. I have to give my Smart ED back, as it is on lease...

Best Regards,
Ralph

casainho, 2017-09-01:
> 9vgaJpZM4NWGoF>.

I bought my Smart ED on a Portuguese Mercedes stand -- do you know how I
can verify the version of my SmartED??

jim-sokoloff, 2017-12-02:
Hi casainho,

When I was building one of these, I got a "workshop-brake fault" message (as well as another error message).

It turned out that one of the 6 pins in the 3x2 pin connector (or maybe one of the side connectors; I'm not 100% sure) was shorted to another pin. In other words, this was a hardware problem, not a software problem.

Check the soldering of those pin headers carefully, especially for bridges. Once I corrected that, the device worked as expected.

casainho, 2017-12-02:
Thanks. I will check but maybe in 2 weeks.