MyLab-odyssey/ED_BMSdiag #18: Ambient temperature

aospan, 2018-09-01:
@MyLab-odyssey, I have found CAN id for ambient temperature. Here is details:

```
Ambient temperature
CAN ID 0x408

Formula:
Temp degC = data[2]/2 - 40

Example:
Temp = 0x8A/2 - 40 = 29 degC

```
Ambient temperature sensor located in front of the car (near radiator). Here is photo from internet:
[image]

it may be useful for you.

aospan, 2018-09-01:
my notes, as bonus :)
[image]

aospan, 2018-09-01:
And here is how my terminal looks like when I'm comparing CAN data in different ambient temperatures. I did 3 CAN-dumps for 28degC, 30degC and 31degC then compared all PID's and looking for some that have changed little bit (Miami has small variation during the day just +-3 degC :).

[image]

MyLab-odyssey, 2018-09-01:
Dear aospan,

great findings! I wanted to publish my complete list of IDs, but have not managed it yet - sorry! But I am really busy these days and wanted to have look over the old data first.

These are my notes for ID 0x408:

```
(counting bit 1 to bit 8)
#408: b3 and b5 Ambient Temperature Fahrenheit ((x-50)-32)*5/9
	b4: Lights ?
	b6, b7: Energy? Counter?
	#when charging it is linear!
	b6*256+b7 Time in minutes? or mean velocity?
```

By the way ID 0x408 is one of the leftovers of the Zytek / Tesla legacy within the ED2 ;-)

I think with the other IDs you can also check the `canDiag.cpp` for live readings...

Cheers, Odyssey

aospan, 2018-09-01:
@MyLab-odyssey, thanks for info !
BTW, your formula gives:

((0x8A - 50) - 32) * 5/9 = 31.1 degC but I have checked dashboard and it shows 29degC when 0x8A. My above formula gives this 29degC. May be it depends on Smart fortwo version (year of manuf., etc) ?

second temperature byte (b5 in your comment) looks like historic data for temperature (previous value). It can be used to estimate what happens outside (is temp rising or falling) - this is just my guess.

MyLab-odyssey, 2018-09-01:
Your version seems really sleek and would give a useful resolution! I never paid that attention to the ambient temp in the live IDs. Let's stick with your version ;-)

aospan, 2018-09-01:
@MyLab-odyssey I have checked now:

```
$ candump can0,7e5:7FF,7ed:7FF
  can0  7E5   [8]  03 22 23 02 FF FF FF FF
  can0  7ED   [8]  05 62 23 02 00 91 00 00

```

and  live (0x408 right ?):
```

$ candump can0,408:7FF
  can0  408   [8]  FF 30 A6 46 A6 0F FE 02
  can0  408   [8]  FF 30 A6 46 A6 0F FE 02
  can0  408   [8]  FF 30 A6 46 A6 0F FE 02
  can0  408   [8]  FF 30 A6 46 A6 0F FE 02
  can0  408   [8]  FF 30 A6 46 A6 0F FE 02

```
(yes, it's 0xA6/2 - 40 = 43 degC - AC is working on parked car is a lot of heat near radiator :)

I don't see correlation between 0x408 and 0x7ED reply.

MyLab-odyssey, 2018-09-01:
...and this is the poll for the RAW ambient temperature:
```
can0  7A2   [8]  02 21 10 FF FF FF FF FF
can0  7A3   [8]  03 61 10 38 00 12 24 03
```
0x38 / 2 = 28 deg C (RAW senor value without correction)

aospan, 2018-09-02:
7a2/7a3 gives me strange results:

```
$ candump can0,7a2:7FF,7a3:7FF
  can0  7A2   [8]  02 21 10 FF FF FF FF FF
  can0  7A3   [8]  03 7F 21 11 00 00 00 00
  can0  7A2   [8]  02 21 11 FF FF FF FF FF
  can0  7A3   [8]  03 7F 21 11 00 00 00 00
  can0  7A2   [8]  02 21 12 FF FF FF FF FF
  can0  7A3   [8]  03 7F 21 11 00 00 00 00

```

MyLab-odyssey, 2018-09-02:
Strange?! Could be a software version problem, as the ECU signals a failure (0x7F) and that the service is not supported (0x11). My ECU did not use standard UDS (ISO 15765) and I suspected it to talk in K-Line via CAN? Maybe they changed the ECU / software to be fully compliant to standard UDS queries (read by identifier 0x22 xx yy).

Just use your finding (ID 0x408), as it seems correct ;-)

P.S.: I am glad they did not change more within the BMS and that my tool is still running on all vehicles ;-)

aospan, 2018-09-02:
I have checked UDS 0x22 (ReadDataByIdentifier):

```
  can0  7A2   [8]  02 22 10 FF FF FF FF FF
  can0  7A3   [8]  03 7F 22 80 00 00 00 00
```

0x80 is ISOSAEReserved | This range of values is reserved for future definition.

seems like not fully compliant to UDS.
Ok, I will try to do more experiment ... :)

jickeydu, 2019-01-07:
Hey Guys,

I am battery engineer also bought this smart ED ，Could you share me the CAN protocol ?

MyLab-odyssey, 2019-01-07:
Sorry, there was and is no time for me to get all info into a CAN matrix. Please refer to the (well documented) code in the canDiag-class. For the queries look into the corresponding header files for the specific ECUs...

closed 2020-09-01 by MyLab-odyssey