MyLab-odyssey/ED_BMSdiag #19: Help needed with Smart Car EV BMS

vioilio, 2018-09-18:
Hello. My Smart Car EV Electric is giving me codes stored in the BMS:
P1D799A - HV battery contactor open due release fault. Ops condition not met.

U01A200, U01A100 and U01A000 - lack of CAN communication with internal control unit with A, B and C cells respectively.

P1614FA -Control unit internal fault initialization unsuccessful.

Car was involved in an accident and airbags were deployed. We have fixed everything but those errors wont clear. Can you please help?

MyLab-odyssey, 2018-09-18:
Seems to be a communication problem of the BMS board with the three cell supervisory electronic boards (CSE). What is the voltage level of the pack? The CSE boards are feed by the module (á 31 cells) itself and are galvanically separated from the 12V bus. My friend Sonny Dahunsi found a special deactivation input on those CSE boards and maybe this was triggered by the accident? But I suspect the HV voltage(s) to be depleted. What is the status of the HVIL? Is the Pyrofuse repaired?

I am sorry, but without access to the pack and without further measurements this is beyond my abilities!

vioilio, 2018-09-18:
Hi. Do you have Viber or whatsup?

vioilio, 2018-09-18:
> Sorry, no other communication channels than this one here! This conversation should be useful also for others: Please tell me the voltage level of the complete pack...

I have no access  to the voltage pack. It says no signal available. Look at this picture
[Image]

MyLab-odyssey, 2018-09-19:
I can only tell you what Sonny did with his "dead" voltage pack:

Initially he had the same fault codes.
[image]
He removed the battery from the car and opened it up. Due to the HV contactors you can't access the terminals (plus/minus) from the outside. Engaging the contactors is overridden by the BMS safety mechanism.
All cells were at about 1V and he bravely tried to charge them with a very low current (about 300 - 350 mA) and closely monitored the process. Only one cell was really bad and the other ones reached acceptable levels. Then the CSE boards had enough power to function and the BMS was able to communicate with them. Now he managed to reset the fault codes, but the defective cell prevented to clear the internal fault status within the BMS and the HV contactors still could not be engaged :(

So for further progress you have to remove the pack and open it up...
[image]

vioilio, 2018-09-19:
I really appreciate all your help. I will remove the battery today. Can you please tell me how to actually charge it? Should i charge the complete battery or i should charge cell by cell. Can you guide me in this process?

[initial bms fault code]<https://user-images.githubusercontent.com/14249372/45737567-7ab39480-bbee-11e8-9aac-2a1667da76b9.JPG>

[sam_0328]<https://user-images.githubusercontent.com/14249372/45739006-227e9180-bbf2-11e8-94d5-d1e696e1c484.JPG>

vioilio, 2018-09-19:
Can you tell me how to charge the battery? Im only getting 30v from one side to the other?
[Image]

________________________________
From: Chanko Penchev <chankopenchev@hotmail.com>
Sent: Wednesday, September 19, 2018 1:39:00 PM
To: Chanko Penchev
Subject: Re: [MyLab-odyssey/ED_BMSdiag] Help needed with Smart Car EV BMS (#19)

Can you show me where is the + and where the - on each cell?
[Image]

________________________________
From: Chanko Penchev <chankopenchev@hotmail.com>
Sent: Wednesday, September 19, 2018 7:03:08 AM
To: MyLab-odyssey/ED_BMSdiag
Subject: Re: [MyLab-odyssey/ED_BMSdiag] Help needed with Smart Car EV BMS (#19)

I really appreciate all your help. I will remove the battery today. Can you please tell me how to actually charge it? Should i charge the complete battery or i should charge cell by cell. Can you guide me in this process?

vioilio, 2018-09-19:
Can you tell me how to charge the battery or the individual cells? Im only getting 30v from the one side to the other.
[Image]

vioilio, 2018-09-19:
Also can you show me on the picture where are the plus and minus terminals on each cell so i can charge them individually?

MyLab-odyssey, 2018-09-19:
Charging a "dead" Li-Ion cell is really dangerous! You have to monitor the current carefully and start with an even lower current (about 50mA) for a first test. See here: <https://www.eevblog.com/forum/beginners/reviving-a-0v-lithium-ion-battery/>

Sonny did charge the pack completely with a 400V DC source set to 350mA max. The cells are connected in series, so you need that voltage at least. His pack hat about 110V initally (when I remember correctly).

With two lab dc power supplies in series you could test the current drain first. Test the current limiting on an resistor (low value) and an ampmeter first. Kown your setup and work carefully!

I will try to find a good picture to give you insight to the cell contacts. The foil cable is very sensitive and contains a thermocouple under each "dot" - DO NOT break them! Everything is FRAGILE there!!!

vioilio, 2018-09-19:
Can i just try to charge each cell individually while there are on the pack?

MyLab-odyssey, 2018-09-19:
I think this would work, as long as you get a good contact to the poles. The [HEA50 cells](https://generationstrom.files.wordpress.com/2018/08/13.jpg?w=736) look like this: and are connected in series by [busbar clamps](https://bilder3.n-tv.de/img/incoming/origs20376734/2152539839-w1280-h960/65514958.jpg). Try to measure the cell voltages of one of the three modules and sum them up and verify with the voltage of that module (31 cells).

MyLab-odyssey, 2018-09-19:
Also see this picture [of the contacts](http://www.dnn.de/var/storage/images/dnn/region/region-news/daimler-stuttgart-will-stueck-vom-kamenzer-kuchen-abhaben/517831838-1-ger-DE/Daimler-Stuttgart-will-Stueck-vom-Kamenzer-Kuchen-abhaben_big_teaser_article.jpg) normally covered by the cooling loops.

vioilio, 2018-09-19:
Can you point me the exact location of the + and - on each cell? The exact points to which i will connect the charger. I also bought this charger: https://www.amazon.com/gp/product/B01MZ1ZZ7Z/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1
Amazon.com: SKYRC iMAX B6AC V2 AC/DC Dual Power Professional LiPo Battery Balance Charger/Discharger: Toys &amp; Games<https://www.amazon.com/gp/product/B01MZ1ZZ7Z/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1>
Buy SKYRC iMAX B6AC V2 AC/DC Dual Power Professional LiPo Battery Balance Charger/Discharger: Office Products - Amazon.com ✓ FREE DELIVERY possible on eligible purchases
www.amazon.com

Do you think it will work?

vioilio, 2018-09-19:
> Theoretically - yes, but you are dealing with 51Ah capacity per cell and this will take way to long. Also getting a stable contact is very difficult on the individual cells! Each module (á 31) has a positive an negative terminal to interconnect to the other module(s). Test the polarity there and disconnect one from the other two modules. Get a lab dc source like this<https://www.amazon.com/Precision-0-100V-Adjustable-Digital-Regulated/dp/B01831QRFY> to charge one module to about 100V resulting in about 3.2V per cell. The CSE should definitely work with with cell voltages of 3V and above (>= 93V on a module). Do you have a good multimeter for verifying the current?

I have multimeter but I do not have a charger. I only have have the one that I showed from amazon you and also 12v 2amp charger. What is your suggestion? How should I proceed?

MyLab-odyssey, 2018-09-19:
Are you familiar with the charger (from amazon)? Get uses to it and its menu system. Test charging on another cell (e.g. from RC toy).

When you start using this charger, you have to charge all cells individually! This is not a good idea! You can use my previous method for each module or get / make a supply with a even higher voltage (Sonny charged to 327V in total). BTW: where are you located? Can someone else help?

I will get you the polarity picture tomorrow...

vioilio, 2018-09-19:
Not familiar with it. I'm located in USA Delaware state. What was your method? How can I make a supply so i can charge each module to 110v?

MyLab-odyssey, 2018-09-19:
Do you know someone with good electronic skills and some equipment? I think you should get some help.
 I recommend a laboratory power supply with about 100V and constant current mode (adjustable) like in the link 
<https://www.amazon.com/Precision-0-100V-Adjustable-Digital-Regulated/dp/B01831QRFY> or this one (1A max): <https://www.amazon.com/Precision-0-100V-Adjustable-Digital-Regulated/dp/B01N4W33EF>

vioilio, 2018-09-19:
So best way would be to charge it at 100v 1amp for 50 hours module by module. I disconect the bus lines and separate each module. Then i connect and charge each module.

MyLab-odyssey, 2018-09-20:
NO, not 1 A at the beginning! Start with <20 mA first for 1min and monitor the voltage levels. When they are rising, slightly get the current up to 50 mA and then carefully increase when the cell voltage is over 2.2 V. Please don't use currents > 350mA before you reach the 2.8V on ALL cells!
This is really tricky and you should know what you are doing! Contact someone who is familiar in reviving run down cells - maybe someone of <https://www.diyelectriccar.com> or <https://www.eevblog.com> is around your Delaware location?

Watch the video in my link (<https://www.eevblog.com/forum/beginners/reviving-a-0v-lithium-ion-battery/>) closely and think of the energy about 20 times higher in one HEA-50 (the cell in the E18-1 Smart battery)!

vioilio, 2018-09-20:
It is very hard to find 100v power supply that has adjustable current. Do you think I can use 60v power supply? I can buy 2 of them and adjust them to 50v and around 20mA each. Then I will connect them in series and start the process. Once i reach 2.8V on each cell I can switch to one power supply of 100v @1A. Do you think this will work or no?

  Take a look at this.

https://www.amazon.com/Haitronic-HPS605D-Adjustable-Switching-Alligator/dp/B072M79X96/ref=pd_sbs_469_2?_encoding=UTF8&pd_rd_i=B072M79X96&pd_rd_r=5986af39-bcc8-11e8-9069-65b0c08f01f2&pd_rd_w=ySRPy&pd_rd_wg=qN63B&pf_rd_i=desktop-dp-sims&pf_rd_m=ATVPDKIKX0DER&pf_rd_p=0bb14103-7f67-4c21-9b0b-31f42dc047e7&pf_rd_r=PFCH2RN3DG1NEM8M6GAE&pf_rd_s=desktop-dp-sims&pf_rd_t=40701&psc=1&refRID=PFCH2RN3DG1NEM8M6GAE

MyLab-odyssey, 2018-09-20:
Connecting two supplies in series will work - but is not the best idea:
when one supply is reaching the current limit, it will reduce / shutdown the voltage to zero. The other one is still active unless it will see the same exact current limit. In addition the units must have a floating output not tight to ground (from mains). Normally DC lab power supplies have this "feature". 

Very good ones like from Keysight or other brand companies have a auto tracking feature, if you connect the units in series, but they are very pricy!

Take a look here for further reading: https://www.eevblog.com/forum/testgear/connecting-2-power-supplies-in-series-for-2-sided-voltage/msg1387022/#msg1387022>

vioilio, 2018-09-20:
My idea is to set both supplies to 50v and 10mA. This will give me total of 100v and 10mA. I will conect to the module of 31 cells and this should be enough to get the cells to 2.8V.
Once this is done and all 31 cells have 2.8V in them I will connect the other power supply of 100v 1A to the module and charge and monitor the process.
Sounds like a plan?

MyLab-odyssey, 2018-09-20:
You can also do it step by step. First use one supply at about 50 to 60 V and 10 mA to get the individual cell voltage to > 1.5 V. In this case you only have to deal with one current regulation and can concentrate on monitoring the voltage of that module. when everything is going well you can use two supplies and rise the voltage and later the current. Always keep it simple...

vioilio, 2018-09-20:
It looks like there is no way to check actual cell voltage without removing the cooling plates and the electronic board. At this point i wont do that. I will just start with 50v and 50mA for 5min. Then i will do 60v 100ma for 5 min. And then i will do 70v 200ma and 80v to 300ma all for 5 min. At the end i will do 100v at 500ma for 5- 10min. After all this will let the pack sit for an hour. Will check the voltage after that and if 90v or more i will do 100v at 1a for about an hour. Is all this correct?

MyLab-odyssey, 2018-09-21:
Getting contact (for multimeter measurement) to the individual cells is really easy when you take off the cover of the CSE board:
[image]

I will advise you only one more time: BE CAREFUL! setting the current only depends on the conditions you actually see when applying current (as low as possible) for the first time. Please set a current limit and carefully rise the voltage from zero to the desired value first. If you produce an inrush current in the first attempt / contact you will ruin the cell(s). DON'T BE IMPATIENT!!! Findig the right procedure will take time...

BUT it is totally up to you! I made my remarks!

MyLab-odyssey, 2018-09-21:
Sorry my fault. The point is really not covered by the CSE cover. But here you will find the very logical arrangement of 31 cells in series:
[image]

vioilio, 2018-09-21:
[Image]

After initial charge of low voltage low current now I'm charging 2 of the 3 modules. Im applying 100v at 0.2A currently.

[e18-1 cell arrangement]<https://user-images.githubusercontent.com/14249372/45889708-6a074800-bdc1-11e8-927b-ed39d00b5d01.png>

bhorrock, 2018-09-21:
I'd like to express my appreciation to you both for the great information that is appearing in this thread.   This is something that MB would never tackle, and for Smart ED 451 owners this is absolutely priceless.

THANK YOU.

MyLab-odyssey, 2018-09-21:
@vioilio: Thumbs up and fingers crossed! Please report. Can you please use github via your browser and upload pictures with the editor there? None of your pictures via e-mail response worked?!

@bhorrock: the credit belongs to my dear friend Sonny Dahunsi. He was brave enough to rip off and dig into the battery and hand over all the details. I am pleased to share his insights here - THX Sonny!

dkraft, 2018-09-21:
100v @ 200ma ? isnt that going to take 29 days ?

MyLab-odyssey, 2018-09-22:
The task is to revive the depleted cells and to do this as carefully as possible. Normally the voltage limit of Li-ion is about 2.5V and levels under this limit lead to significant degradation. The voltage slope is really steep in this area and the low currents have to be applied there. I would suggest to increase the current when passing 2.8V (Open Cell Voltage = without current and after settlement) and increase to <= 1A. After passing 3V the current maybe increase further but not monitor the cell temperatures carefully, as the internal resistance may be significant due to the degradation process. Sonny did find a defective cell not holding the charge and getting "hot" because of this process. The other cells could be charged with 6A but we don't know what is happening when driving currents >100A are used?

Here the OCV diagram:
[image]

MyLab-odyssey, 2018-10-08:
@vioilio: any update?! Please tell us about your progress...

nsb002, 2019-01-13:
Hello, I don't know if you can help me and @MBTP with this fault code when the 12V battery dies :
HV System → Workshop
[image]

I don't know if you remember us from another post about the CANBUS address to set the departure time? It works well now, we added a Sonoff to start the car remotely and switched to an Arduino Mega 2560 to be able to add some code and functionalities.
You probably know how to remove this fault code? I have a Bluetooth OBD II fault reader and it doesn't work at all on the Smart EV.

Thanks in advance.

MBTP, 2019-01-14:
Please help me for resetting message HV System workshop. If I drop 12v battery to low this message appears. I use sonoff remote for programming departure time with Arduino mega. This control is full functionality for programming departure time am and pm automatically

MyLab-odyssey, 2019-01-14:
Please fully describe your problem! As far as I understand you get the HV error when the 12V battery is depleted (e.g. due to very low temperatures). Normally the error is due to a timing failure when the HV-contactors get not enough current from the 12V battery and they don't get into the on position in the right time. Recharging the 12V battery will normally clear the fault in the next attempt. What is the voltage of the 12V batter?
If you know that the system is well you have to wait at least until the "P" indicator will switch off - lock the car and wir for 5min and then give it another try.
If the failure persists you may have a damaged HV contactor - what is the actual cycle count? - or you have an insulation problem. Please use my tool and show a complete readout (with 'all' command).

nsb002, 2019-01-16:
The car works well. In fact, I already went two times at the dealers mechanic garage to have this fault erased, the problem is that it costs 100$ each time, so I would be happy to have a free alternative to erase this kind of fault code. Every time it's the same story: the 12V battery died due to very low temperature and because I have a lot of electronics powered at all time (Arduino, Sonoff, Cell Phone for WiFi and Android Radio). I also already replaced the battery less than a year ago. Is there a way to clear the messages with Arduino code and the same CANBUS interface? This fault code is really annoying because I cannot use the dashboard controls to set the time or the departure time.

### 'all' command readout:
```
Reading data....................OK
-----------------------------------------
Time [hh:mm]: 03:24, ODO : 103399 km
-----------------------------------------
Battery Status : OK

Battery Production [Y/M/D]: 2014/2/3
Battery-FAT date [Y/M/D]: 2014/2/3
Rev.[Y/WK/PL] HW:2013/29/0, SW:2014/18/0
-----------------------------------------
SOC : 43.0 %, realSOC: 62.5 %
HV : 321.1 V, -41.44 A, -13.31 kW
LV : 14.9 V
-----------------------------------------
CV mean : 3441 mV, dV = 128 mV
CV min : 3339 mV
CV max : 3467 mV
OCVtimer: 62593 s
-----------------------------------------
Last measurement : 3 day(s)
Measurement estimation: 0.627
Actual estimation : 0.560
CAP mean: 17049 As/10, 47.4 Ah
CAP min : 14755 As/10, 41.0 Ah
CAP max : 18139 As/10, 50.4 Ah
-----------------------------------------
HV contactor state ON, low current: 0 s
Cycles left : 283523
of max. cycles: 300000
DC isolation : 65534 kOhm, OK
-----------------------------------------
Temperatures Battery-Unit /degC: 
module 1: 1.1, 7.8, 3.1
module 2: 2.1, 7.9, 3.7
module 3: 0.8, 7.7, 2.4
mean : 4.1, min : 0.8, max : 7.9
coolant : -6.3
-----------------------------------------
# ;mV ;As/10
01;3340;12952
02;3406;16074
03;3419;16532
04;3427;16802
05;3436;17081
06;3438;17175
07;3442;17336
08;3445;17207
09;3429;17175
10;3450;17434
11;3453;17401
12;3455;17533
13;3454;17434
14;3456;17533
15;3456;17401
16;3458;17533
17;3457;17434
18;3459;17336
19;3456;17368
20;3456;17401
21;3455;17467
22;3454;17336
23;3451;17336
24;3451;17368
25;3446;17271
26;3444;17239
27;3441;17018
28;3438;17175
29;3434;17049
30;3418;16415
31;3369;14340
32;3379;16018
33;3420;16956
34;3429;16987
35;3438;17144
36;3445;17271
37;3449;17239
38;3453;17336
39;3453;17368
40;3458;17336
41;3461;17368
42;3463;17401
43;3465;17368
44;3463;17401
45;3463;17336
46;3466;17467
47;3464;17336
48;3466;17336
49;3465;17336
50;3466;17304
51;3464;17304
52;3463;17336
53;3464;17336
54;3462;17336
55;3460;17239
56;3458;17112
57;3455;17081
58;3455;17434
59;3450;17304
60;3444;17175
61;3433;16987
62;3392;16474
63;3360;14871
64;3405;16532
65;3420;16956
66;3428;17239
67;3430;17271
68;3437;17467
69;3440;17434
70;3437;17271
71;3440;17207
72;3444;17336
73;3445;17271
74;3450;17336
75;3449;17336
76;3450;17271
77;3451;17336
78;3452;17304
79;3452;17401
80;3451;17401
81;3453;17239
82;3450;17271
83;3451;17144
84;3451;17304
85;3449;17304
86;3446;17175
87;3444;17271
88;3442;17112
89;3436;17049
90;3433;16987
91;3425;16894
92;3423;17207
93;3365;14908
-----------------------------------------
Individual Cell Statistics:
-----------------------------------------
CV mean : 3442 mV, dV= 126 mV, s= 23.95 mV
CV min : 3340 mV, # 1
CV max : 3466 mV, # 46
-----------------------------------------
CAP mean: 17083 As/10, 47.5 Ah
CAP min : 12952 As/10, 36.0 Ah, # 1
CAP max : 17533 As/10, 48.7 Ah, # 12
-----------------------------------------
Voltage Distribution (dV= 126 mV):
*------------------->---------[===|=]---*
3340 06 > [3436; 3450; 3456] < 00 3466
min [p25; median; p75] max
-----------------------------------------
Reading data....OK
-----------------------------------------
Status OBL Charger-Unit: 
User selected : 16 A
Cable maximum : 204 A
AC L1: 0.0 V, 0.0 A
DC HV: 0.0 V, 0.0 A
DC LV: 14.9 V
-----------------------------------------
Temperatures Charger-Unit /degC: 
Reported : NA
Cooling plate : NA
Inlet socket : NA
-----------------------------------------
Reading data.OK
-----------------------------------------
Status Cooling- and Subsystems: 
Temperature : -14.5 degC
Cooling fan : 0.0 %
Cooling pump : 57.3 %, -12 degC
: 14.4 V, 1.2 A
OTR:
Cooling fan : 204 h
Cooling pump : 10317 h
Battery heater: 72 h, OFF
Vaccum pump : 65.728 h
Pressure 1, 2 : -713 mbar, -713 mbar
-----------------------------------------
CMD >>
```

minorukurata, 2019-02-01:
Hi I also have my smart fortwo ED 2014 with before that with BMS error P18051c I already fix the issue with softelectronic and charged the HV battery but right now Icant turn on the car

eading data....................OK
-----------------------------------------
Battery VIN: WMEEJ9AA2EK736711
Time [hh:mm]: 12:00,   ODO : 11289 km
-----------------------------------------
Battery Status   : FAULT

Battery Production [Y/M/D]: 2013/8/19
Battery-FAT date   [Y/M/D]: 2013/8/19
Rev.[Y/WK/PL] HW:2013/29/0, SW:2013/29/1
-----------------------------------------
SOC : 84.5 %, realSOC: 81.5 %
HV  : 369.1 V, -1024.00 A, -10.24 kW
LV  : 12.1 V
-----------------------------------------
CV mean : 3968 mV, dV = 272 mV
CV min  : 3870 mV
CV max  : 4142 mV
OCVtimer: 1075 s
-----------------------------------------
Last measurement      : 254 day(s)
Measurement estimation: 0.792
Actual estimation     : 0.083
CAP mean: 18333 As/10, 50.9 Ah
CAP min : 18514 As/10, 51.4 Ah
CAP max : 18777 As/10, 52.2 Ah
-----------------------------------------
HV contactor state OFF, for: 0 s
Cycles left   : 296478
of max. cycles: 300000
DC isolation  : 14060 kOhm, OK
-----------------------------------------
Temperatures Battery-Unit /degC: 
module 1: 33.3, 32.9, 34.5
module 2: 33.5, 32.9, 33.5
module 3: 34.4, 33.3, 34.1
   mean : 33.6, min : 32.9, max : 34.5
coolant : 36.0
-----------------------------------------
# ;mV  ;As/10
01;4109;18212
02;3877;18152
03;4077;18212
04;3879;18182
05;4030;18122
06;3879;18152
07;4118;18092
08;3879;18062
09;4021;18062
10;3880;18242
11;4028;18182
12;3878;18152
13;4023;18092
14;3877;18182
15;4137;18092
16;3880;18032
17;4077;18152
18;3880;18212
19;4087;18152
20;3880;18212
21;4026;18242
22;3873;18212
23;4146;18212
24;3873;18212
25;4064;18333
26;3867;18182
27;4030;18242
28;3879;18242
29;4017;18272
30;3874;18092
31;4018;18182
32;4024;18182
33;3873;18062
34;4027;18212
35;3867;18122
36;4052;18122
37;3869;18122
38;4084;18182
39;3881;18092
40;3977;18152
41;3877;18062
42;4005;18062
43;3879;18152
44;4019;18032
45;3880;18032
46;4010;18032
47;3879;18092
48;4134;18152
49;3878;18182
50;4034;18152
51;3878;18092
52;4028;18122
53;3878;18062
54;4065;18032
55;3877;18032
56;4034;18032
57;3875;18032
58;4043;18062
59;3872;18032
60;4025;18062
61;3877;18062
62;4038;18003
63;4048;18242
64;3881;18182
65;4081;18122
66;3884;18242
67;4018;18303
68;3884;18182
69;4030;18303
70;3886;18303
71;4030;18242
72;3885;18272
73;4140;18272
74;3883;18242
75;4046;18242
76;3882;18212
77;4025;18032
78;3877;18092
79;4066;18032
80;3878;18303
81;4043;18182
82;3880;18092
83;4110;18092
84;3879;18092
85;4048;18062
86;3876;18272
87;4080;18242
88;3880;18333
89;4095;18333
90;3880;18272
91;4045;18152
92;3877;18092
93;4028;18092
-----------------------------------------
Individual Cell Statistics:
-----------------------------------------
CV mean : 3968 mV, dV= 279 mV, s= 91.89 mV
CV min  : 3867 mV, # 26
CV max  : 4146 mV, # 23
-----------------------------------------
CAP mean: 18154 As/10, 50.4 Ah
CAP min : 18003 As/10, 50.0 Ah, # 62
CAP max : 18333 As/10, 50.9 Ah, # 25
-----------------------------------------
Voltage Distribution (dV= 279 mV):
*[==================|====]--------------*
3867   00 > [3879; 4007; 4043] < 00  4146
min         [p25; median; p75]        max
-----------------------------------------
Reading data....OK
-----------------------------------------
Status OBL Charger-Unit: 
User selected : 32 A
Cable maximum : 204 A
AC L1: 0.0 V, 0.0 A
DC HV: 0.0 V, 0.0 A
DC LV: 12.3 V
-----------------------------------------
Temperatures Charger-Unit /degC: 
Reported       : NA
Cooling plate  : NA
Inlet socket   : NA
-----------------------------------------
Reading data.OK
-----------------------------------------
Status Cooling- and Subsystems: 
Temperature   : 38.2 degC
Cooling fan   : 0.0 %
Cooling pump  : 80.0 %, 40 degC
              : 11.9 V, 0.2 A
OTR:
Cooling fan   : 14 h
Cooling pump  : 1838 h
Battery heater: 0 h, OFF
Vaccum pump   : 21.797 h
Pressure 1, 2 : -668 mbar, -665 mbar
-----------------------------------------
CMD >>

also have some error 

I didt try yet MB star scanner only with Launch x431 pro

[image]

I cant find information about P180D1D and the other one. please if @MyLab-odyssey @vioilio vioiliocan give me a hand
regards

vioilio, 2019-02-03:
Hi everyone. Its been a while since my last time here. I have been extremely busy the last couple of months and have not had a chance to give any updates. 
Charging my battery was success! The car is currently getting used on a dally basis and no problems so far. 
To answer some of the questions here: 
All users above that asked for help with "HV Workshop Message". You must have Star Diag Mercedes diagnostic system. You must clear all errors and you must run software restart on all modules. If both of the batteries are charged this will take care of any residual problems after charging the HV battery. 
It is the same for user "minorukurata". Clear all error codes with Star DAS and restart the software on all modules. If both of the batteries are charged and you do not have error code P18051c you should be able to start and drive your car.

minorukurata, 2019-02-03:
Thanks @vioilio for answering i will try tomorrow. Both batteries are charged will keep update

vioilio, 2019-02-04:
Have you scan the car with Star Das?

minorukurata, 2019-02-04:
today I will, only have Launch X431 Pro with smart 17.90 has the electric version updated, was trying if it erase the dtc airbag or some trouble it does, but no with N82/2 HV battery

minorukurata, 2019-02-04:
I contacted with a mechanic that he has will come here to see, I doubt, let see, Im guessing something wrong inside the HV

fxmatar, 2019-02-14:
Any updates did you find a way to solve this issue about sensor A code, as I’m facing the same.
Can any body help please.
@minorukurata

minorukurata, 2019-02-14:
hi @fxmatar  Im working on I found a supple of sensor A 
was having a 1 o 2 flex cutted, I got  from a salvage vehicle from a friend in oregon, his is sending my to Paraguay, but the BMS bricked again with P18051c, I sent again to fix it. mine was

minorukurata, 2019-02-25:
@vioilio did your car get the P18051c error DTC?

minorukurata, 2019-02-25:
@vioilio I received another smart fortwo electric 2013 with same errors, I found the WHOLE HV battery Pack at 21V, module 1 (31S) has a lot of cell doest up voltage, module 2 (31s) also, only module 3 is charging upto 120VDC, I found with 0 v a lot of cell how did u do for charging ?

minorukurata, 2019-04-25:
@vioilio @MyLab-odyssey I found something rare, when I measure with multimeter each cell are at 3.99 - 4.07v max but when I scan with the board shows me diferent voltage 

I got another car with whole sensor A B and C intact, and welderd into my pack because the pack in the car I took the sensor had a lot of Cell damaged more than 20 so I decided to use the sensor into my battery.  I found that the measuring voltage from the BMS (scanner) MB star or Launch is more diffent that measuring with multimeter. some mechanica in spain with experience told me is because the Delta volt is too high the battery can balance itself, and I have to discharge down below 5% and charge again
# ;mV  ;As/10
01;4048;18078
02;3835;18049
03;3991;17588
04;3837;17532
05;4011;17616
06;3837;18228
07;4057;17758
08;3772;18228
09;3984;18049
10;3772;18138
11;4019;18168
12;3819;18138
13;3982;17644
14;3818;17588
15;4013;17560
16;3823;18198
17;3988;18258
18;3818;17701
19;4013;18019
20;3817;18019
21;4002;18019
22;3791;17532
23;3998;18049
24;3792;17729
25;4017;17701
26;3787;17588
27;4007;17701
28;3732;17588
29;3997;17532
30;3770;17729
31;4038;17729
32;3963;18078
33;3852;18318
34;3973;18288
35;3847;18108
36;3990;17787
37;3848;18318
38;4019;17990
39;3859;18019
40;3922;18108
41;3857;18168
42;3947;17990
43;3859;17560
44;3961;17504
45;3859;18138
46;3952;18198
47;3859;17673
48;4008;17532
49;3856;17532
50;3971;17644
51;3856;17644
52;3964;18108
53;3856;18198
54;4003;17644
55;3855;17532
56;3973;18019
57;3854;17729
58;3984;17931
59;3854;17616
60;3966;18288
61;3858;18410
62;4029;17729
63;3995;18168
64;3861;18138
65;4027;18318
66;3864;18198
67;3964;18198
68;3864;18787
69;3973;18318
70;3862;18258
71;3974;18258
72;3863;18108
73;4065;18108
74;3862;18108
75;3991;18138
76;3861;18318
77;3964;18138
78;3856;18659
79;4005;18288
80;3857;18198
81;3988;18078
82;3860;18198
83;4043;18078
84;3859;18441
85;3987;18288
86;3856;18108
87;4019;18288
88;3859;18228
89;4031;18228
90;3859;18228
91;3986;18198
92;3857;18503
93;3986;18228
-----------------------------------------
Individual Cell Statistics:
-----------------------------------------
CV mean : 3920 mV, dV= 333 mV, s= 84.07 mV
CV min  : 3732 mV, # 28
CV max  : 4065 mV, # 73
-----------------------------------------
CAP mean: 18011 As/10, 50.0 Ah
CAP min : 17504 As/10, 48.6 Ah, # 44
CAP max : 18787 As/10, 52.2 Ah, # 68
-----------------------------------------
Im very fustrating becuase to get parts of smart Electric is so so so hard .. if you can get me a hand with this

regards

tbnz21, 2019-04-27:
Anyone found solutions for sensor A , 
I am facing the same code with my fortwo smart , and i didn’t find the sensor in dealership, 
Can anyone help !! 
😥
@MyLab-odyssey 
@minorukurata
@fxmatar 
??!

minorukurata, 2019-04-29:
I got another smart ed and took off the whole sensor A, B and C, I gave me the same error. Im really dont know what to do... I already balaced the whole pack to 4.0v

minorukurata, 2019-04-29:
@tbnz21 Im 99.9% is not the sensor I changed a lot even with another smart never touched with a lot of bad cell I welded carefuly but still that error

minorukurata, 2019-04-29:
@MyLab-odyssey  how can I contact with your friend Sonny Dahunsi. ? I want to know about an error Im having.

tbnz21, 2019-04-30:
I solved it, 
I have replaced sensor in the battery with another one , bring it from other smart , and my smart works, 
But how can i put photo here to show you the sensor photo?
@minorukata

minorukurata, 2019-04-30:
@tbnz21 can you share me at minorukun@gmail.com please. I will put the sensor I got 
[image]

minorukurata, 2019-04-30:
@tbnz21 and another thing wich one is the sensor A,  I dont know exactly 
[image]

tbnz21, 2019-04-30:
This is its location

minorukurata, 2019-05-02:
@tbnz21 thanks to Tariq he discovered that the sensor A from Fiat 500e is valid also working 
[image] this sensor is broken in both of my batteries, Im finding out replacement

minorukurata, 2019-05-02:
according to @tb
[image] 
nz21 is alos tried different sensor from 12v batteries uses in honda,  but this sensor doesnt work on smart

tbnz21, 2019-05-02:
Yes 
The numbers H.W and S.W is important to know if the sensor is working with smart battery  
This is my e-mail : tariq.ziad84@gmail.com

dwlimf, 2019-06-21:
Just bought at auction 2013 Smart ED with 20k.
From the pictures HV battery is dead and they jumped the 12 to get the dash to show the miles.
I called the battery assurance number and the extremely helpful lady informed me of the following.
-The car was actually a second owner lease return.
-It was just returned to the dealer last month the end of May.
-The previous owner did have and maintain the battery warranty.
-As what she shows the car is eligible for the warranty to be continued @$80 per month.
-The previous owner did keep up with the scheduled maintenance so if it ever needed a HV battery it would be eligible (if I chose to continue it that is).
-Battery maintenance and service was just done at the dealer in March of this year.

The car is 1200 miles away and I am working on transport seems there is a chance I will get it by Monday.

Out of curiosity I asked if I don't pay for the warranty what happens does MB disable the battery as I have read. She said no they cannot do that you are just out of luck for service and warranty.

My main questions that I cannot see to find conclusive answers to is:

-If the HV is totally dead will it take a charge (seems only a few weeks it would have been dead just at the auction).
-When I get the car what steps should I take when I get the car to have the best chance to get it going again.

She did note that if the battery was run all the way dead and or does not charge that even if I take the battery plan @$80pm they still most likely will not honor the replacement due to running dead is abuse. She said it is really up to the dealer but there is no guarantee.

Just looking for a starting point as there is a lot of conflicting info out there and this thread seems to have some knowledgeable ppl.

Thanks in advance!

fredoo5, 2019-07-04:
I have a 2014 smart electric does anybody have a cse board thats good out of a battery

[20 more replies]

closed 2020-09-01 by MyLab-odyssey