MyLab-odyssey/ED_BMSdiag #22: Compilation error

tamingdigital, 2018-11-10:
Hi,

Thanks a lot for this work. I want to try your project on my Smart ED.
I have the same problem described in this thread : 
[http://clubsmartcar.com/index.php?/topic/28328-charging-issue/&page=2](url)

I have the same idea than _albiesmart_ and maybe your project can help us to clear some battery default...

When i try to compile, i have an error : 

_ED_BMSdiag:306:23: error: variable or field 'ReadGlobalConfig' declared void
 void ReadGlobalConfig(deviceStatus_t *config, bool force_write)
                       ^
ED_BMSdiag:306:23: error: 'deviceStatus_t' was not declared in this scope
ED_BMSdiag:306:39: error: 'config' was not declared in this scope
 void ReadGlobalConfig(deviceStatus_t *config, bool force_write)
                                       ^
ED_BMSdiag:306:47: error: expected primary-expression before 'bool'
 void ReadGlobalConfig(deviceStatus_t *config, bool force_write)
                                               ^
exit status 1
variable or field 'ReadGlobalConfig' declared void_

Can you help me ?

Thanks a lot.

Stephane

MyLab-odyssey, 2018-11-10:
Sorry, but please read the issue section first, before posting a new one! It's a fault within Arduino IDE 1.8.7 builder and can be resolved as described in [Issue#21](
https://github.com/MyLab-odyssey/ED_BMSdiag/issues/21). Use Arduino IDE 1.8.5 or swap the builder...

closed 2018-11-10 by MyLab-odyssey

tamingdigital, 2018-11-10:
Update : 
I try with Arduino 1.8.4 ==> it's working