MyLab-odyssey/ED_BMSdiag #23: HV Battery problem

dgn555, 2019-02-28:


dgn555, 2019-02-28:
I have this errors

U01a2FA  can communication with the internal control unit for monitoring the battery cells (C) is fault
P14f300 The internal function of the high voltage battery module have been deactivated due to cell voltage being love
P0a0e00 The interlock of the high voltage on-board electrical system has a aporadic malfunction
P0de6fb the cell voltage of high voltage battery module are too low
P0de6fa the cell voltage of high voltage battery module are too low
P18051C the hardware monitoring function has detected fault in the control unit. The voltage value is outside the stored permissible range 
here is my logs

Is someone can help me? what must i do? only charging cells one by one is enough to fix?

Connect to OBD port - Waiting for CAN-Bus 
.................CONNECTED
-----------------------------------------
........Reading data....................OK
-----------------------------------------
Battery VIN: WME4513901K641612
Time [hh:mm]: 09:43,   ODO : 7834 km
-----------------------------------------
Battery Status   : FAULT

Battery Production [Y/M/D]: 2012/9/14
Battery-FAT date   [Y/M/D]: 2012/9/14
Rev.[Y/WK/PL] HW:2012/21/0, SW:2014/18/0
-----------------------------------------
SOC : 0.0 %, realSOC: 2.8 %
HV  : 316.7 V, 0.00 A, 0.00 kW
LV  : 12.1 V
-----------------------------------------
CV mean : 3404 mV, dV = 1141 mV
CV min  : 2310 mV
CV max  : 3451 mV
OCVtimer: 4269 s
-----------------------------------------
Last measurement      : 65 day(s)
Measurement estimation: 0.765
Actual estimation     : 0.242
CAP mean: 19081 As/10, 53.0 Ah
CAP min : 18145 As/10, 50.4 Ah
CAP max : 18573 As/10, 51.6 Ah
-----------------------------------------
HV contactor state OFF, for: 0 s
Cycles left   : 193493
of max. cycles: 300000
DC isolation  : 7846 kOhm, OK
-----------------------------------------
Temperatures Battery-Unit /degC: 
module 1: 16.7, 16.5, 16.6
module 2: 16.7, 16.4, 16.8
module 3: 16.5, 16.6, 16.7
   mean : 16.6, min : 16.4, max : 16.8
coolant : 16.3
-----------------------------------------
# ;mV  ;As/10
01;3410;18689
02;3410;18650
03;3419;18766
04;3426;18883
05;3397;18650
06;3397;18650
07;3428;18922
08;3430;19002
09;3431;18883
10;3439;19041
11;3441;19002
12;3013;18727
13;3400;18612
14;3438;19202
15;3405;18650
16;3401;18689
17;3401;18727
18;3428;18844
19;3435;19002
20;3394;18805
21;3439;19081
22;3432;19162
23;3439;19162
24;3396;18612
25;3403;18574
26;3414;18805
27;3416;18805
28;3432;18962
29;3418;18650
30;3432;18844
31;3433;18805
32;3432;18766
33;3428;18805
34;3441;19041
35;3441;19081
36;3431;19002
37;3430;18844
38;3442;19081
39;3439;18883
40;3439;19081
41;3441;19162
42;3435;18844
43;3429;19041
44;3440;19202
45;3443;19243
46;3426;19002
47;3441;19122
48;3433;19041
49;3440;19162
50;3432;19002
51;3441;19202
52;3437;19081
53;3435;19081
54;3436;19081
55;3441;19041
56;3441;19002
57;3439;19041
58;3401;18574
59;3439;19002
60;3439;19081
61;3442;18766
62;3447;18883
63;3453;18883
64;3421;18689
65;3432;18766
66;3420;18766
67;3423;18727
68;3423;18766
69;3429;18805
70;3423;18805
71;3410;18727
72;3419;18689
73;3416;18650
74;3421;18844
75;3423;18844
76;3421;18805
77;3423;18766
78;3422;18844
79;3403;18727
80;3421;18805
81;3415;18844
82;3421;18805
83;2285;18727
84;3415;18805
85;3231;18727
86;3275;18766
87;3399;18727
88;3407;18689
89;3432;19041
90;3411;18727
91;3404;18727
92;3409;18727
93;3424;18574
-----------------------------------------
Individual Cell Statistics:
-----------------------------------------
CV mean : 3405 mV, dV= 1168 mV, s= 127.41 mV
CV min  : 2285 mV, # 83
CV max  : 3453 mV, # 63
-----------------------------------------
CAP mean: 18870 As/10, 52.4 Ah
CAP min : 18574 As/10, 51.6 Ah, # 25
CAP max : 19243 As/10, 53.5 Ah, # 45
-----------------------------------------
Voltage Distribution (dV= 1168 mV):
*------------------------------------>[|*
2285   03 > [3414; 3428; 3438] < 00  3453
min         [p25; median; p75]        max
-----------------------------------------
Reading data....OK
-----------------------------------------
Status OBL Charger-Unit: 
User selected : 32 A
Cable maximum : 0 A
AC L1: 0.0 V, 0.0 A
DC HV: 11.4 V, 0.0 A
DC LV: 12.3 V
-----------------------------------------
Temperatures Charger-Unit /degC: 
Reported       : 14
Cooling plate  : 14
Inlet socket   : NA
-----------------------------------------
Reading data.OK
-----------------------------------------
Status Cooling- and Subsystems: 
Temperature   : 13.5 degC
Cooling fan   : 0.0 %
Cooling pump  : 81.2 %, 14 degC
              : 12.0 V, 0.2 A
OTR:
Cooling fan   : 5 h
Cooling pump  : 738 h
Battery heater: 0 h, OFF
Vaccum pump   : 23.924 h
Pressure 1, 2 : -698 mbar, -701 mbar
-----------------------------------------

-----------------------------------------
Battery VIN: WME4513901K641612
Time [hh:mm]: 09:43,   ODO : 7834 km
-----------------------------------------
SOC : 0.0 %, realSOC: 2.8 %
HV  : 316.7 V, 0.00 A, 0.00 kW
LV  : 12.1 V
-----------------------------------------
ENTER command... (? for help)
CMD >>

minorukurata, 2019-02-28:
the error P18051c U cannot erase it, there is same cell 

12;3013;18727
83;2285;18727

u must check this

dgn555, 2019-03-01:
I am charging all individual cells to 3500 mvolt, shall i need to do anything else?

minorukurata, 2019-03-01:
charge up to 4.1 volt there is a posibility that the cell is weak so is unbalance the entire pack

another thing is the DTC P18051c cannot erase it only I know a company called softelectronic it does

dgn555, 2019-03-01:
May i learn the amperage of the batteries please, for individual cell? 4.1 volt and how much amper?

minorukurata, 2019-03-01:
normal AH of each cell is 53AH,

dgn555, 2019-03-01:
i limited it with 15A is it will be safe or not?

minorukurata, 2019-03-01:
capacity is 53AH you can charge normally at 1C = 53A so the internal charger is 3.3kw so its charge at 10A

closed 2020-09-01 by MyLab-odyssey