MyLab-odyssey/ED_BMSdiag #24: 0x508 in canDiag.cpp possibly incorrect?

mcowher, 2019-03-07:
In comparing the use of your code in the OVMS project, I'm thinking maybe the High Voltage Amps (0x508) is half of what it should be.

I've monitored the affects of charging my Smart EV with a 16A charger.  When the dashboard setting is "Max", the results are coming back at 8A.  When I lower the max charge rate to 8A, the results come back as 4A.  

Just wondering if we need to double your calculations from 0x508?

MyLab-odyssey, 2019-03-07:
Welcome _mcowher_. Please note that the dashboard setting will affect the input current (primary) of the charger. The reading of ID 0x508 is the current measured directly at the battery and is not equal to the input current of the charger from mains. The power distribution chain is mains (primary) -> charger out (secondary) -> battery (with some losses of auxiliary loads as pumps, drivers, ECUs). As the voltave levels of each stage differs, you will get other currents related to the actual power level.

We did a lot of testing and the math for ID 0x508 was also compared to the direct readout from the BMS (see rqBattAmps in _BMS_dfs.h). I will not actively support the OVMS and if you feel comfortable with your findings, use them...

Best Regards
odyssey

closed 2019-03-07 by MyLab-odyssey

mcowher, 2019-03-07:
Thank you for your quick reply Odyssey and thanks for your finding of all these CanIDs!

mcowher, 2019-03-08:
A light bulb just went off for me.  Because the battery volts are in the 380V range, and the battery amps (in my case) are around 8A, that gives me around 3kW of charging rate, which is exactly what I'm expecting when I look at the charging rate reported by commercial charging stations like ChargePoint.

16A/220V/3.52kW coming into the car ends up being 8A/380V/3.04kW at the battery.

Are my assumptions correct?