MyLab-odyssey/ED_BMSdiag #25: 2013 Smart EV HV battery bricked ... after sitting connected to a level 2 charger for 6mo

pilotjono, 2019-04-19:
My car had been parked in a garage, hooked up to a level 2 charger, for 6 months.  The Siemens charger indicated that the car was fully charged.  When I turned the key, nothing happened.  I found that the 12 battery was flat, and recharged it with an external charger.  Everything worked then but the HV battery still showed zero charge.  I called the closest Mercedes dealership authorized to work on these cars, and was told I had to trailer it in.  I related my story to them & left the car.  Two days later they told me that the high voltage battery would not take a charge, and was "dead."  When I pressed, they said that several of the cells in the battery pack were below normal, but the whole pack would need to be replaced  They quoted a price of $9000, and said that the replacement was from 3-6 months away.  I'm wondering if there is an easy way to get to the main DC terminals, and trickle charge the whole pack.

minorukurata, 2019-04-22:
welcome to daimler bms software issues, the HV battery bricked because the voltage get down below 3.4v I guess, Im in process to fix the same issue as u had.

minorukurata, 2019-04-22:
u must get down the battery and charge, the problem is if your bms has P18051c issue, let me know if you can scan your car with MB star c4 o Launch X431 pro

pilotjono, 2019-04-22:
The MB folks hooked up a scanner that reported several cells in the battery were below acceptable levels ... and that this would prevent the car from charging because of the safety system in the BMS.  They offered no explanation about how the battery could go from perfectly fine to dead while the car was simply parked for an extended period and connected to an approved Level 2 charger.  They apparently have no way to verify that the internal charger or BMS are actually functioning properly, and although they indicated that a $9K battery replacement would fix this car, I have zero confidence in their ability to correctly predict this outcome.  I've seen several YouTube examples where this kind of battery was trickle charged back to a point where the on-board charger would fully charge the battery, but in every case, this requires removal of the battery, opening its case, and charging individual battery packs with an external, current controlled HV charger.  If MB can't do better than ring their hands, I'm going to try this.

dkraft, 2019-04-22:
video everything!

nofuel1, 2019-05-06:
Hallo you fixed the problems whit P18051c?

closed 2020-09-01 by MyLab-odyssey