MyLab-odyssey/ED_BMSdiag #33: Troubleshooting and Tips

nighteagle1974, 2023-11-02:
On the Wiki this:
"Load the CAN_receive.ino file (from Arduino > libraries > MCP_CAN_lib_15 > examples) and run it."

Must be chanfge to this:

Load the CAN_send.ino file (from Arduino > libraries > MCP_CAN_lib_15 > examples) and run it.

Then the Start of the Tool is not the same as described - The Message after ".CONNECTED" does not appear.
After some seconds F#0F#1 and so on appear then Reading Data Message appear...  after a few minute the Reading data end with Measurements failed and this two times.

Is this right? Notice i test it without the car - so the Reading start by itself after pwoer Up the Tool?

which behaviour is right?

MyLab-odyssey, 2023-11-03:
This is only a procedure to test the configuration of the Adruino and the CAN shield. Do as described! You have to upload the "CAN_receive.ino" file to the arduino. This is an provided example from the library. I will give no further support - sorry!

closed 2023-11-03 by MyLab-odyssey

nighteagle1974, 2023-11-03:
No - you have to load the CAN_send.ino - because the CAN_receive have not such message in the code!

MyLab-odyssey, 2023-11-03:
THX - BUT: **You are quite late to the show** and none of the cited code does have a .CONNECTED message. Both examples (receive and send) of the provided CAN-lib contain a message, when the CAN-Controller is not initialized by a false CS ("Error Initializing MCP2515..."). If you have a wiring problem with the OBD-cable there are also numerous details about the possible issues.
I WON'T change anything to the documentation! The tool and documentation was succesfully used a lot in the past.
> Just go ahead and live with it as it is...